turing-machines/BMC-Firmware #232: Improvements

gjrtimmer, 2025-01-26:
This issue is to track several improvements.

- [ ] SDCard Image to run from SDCard
- [ ] Package manager #150 
- [ ] Python3 #160 

The improvements are to rename the current SDCard image to `*-sdc.img`, indicating that the image can be installed from the SDCard.

Add a new SDCard image named `*-sdcx.img`, indicating an image that can `run` off the SDCard.
Add a package manager, and doing so also allows for installing Python3 for those who need Python3 for Ansible or other purposes.

I intend to create a PR soon with the first changes to the bmc-firmware build process.

svenrademakers, 2025-01-27:
Hi @gjrtimmer perhaps i can help setting up an Community Edition sd card build here on github? i think this feature is long time due..

gjrtimmer, 2025-01-28:
THat would be a great addition, next to the official builds, so we are taking a `ce` eidtion right, we would still keep the images like tpu, sdc and sdcx

gjrtimmer, 2025-01-28:
Regarding the CE edition, I also mentioned this just now in PR #160 if we add only the package manager and allow the install of package on the SDCard as the install location then users have enough storage. This way we do not have to include Python at all.

PR #160 should be placed `On Hold` while this PR is being worked on.

gjrtimmer, 2025-01-28:
After some additional thinking I think that a community edition would be a great asset, however on priority it would be great to do have automated firmware testing first to accelerate the development. I will also go through all the issues, comment them and see if some can be closed and try to determine the work required for it. Furthermore I think it would be great to create some `Projects` in the Turing Machine organization to handle releases and the issue tracking across multiple repositories. But this is just my opinion.