turing-machines/BMC-Firmware #241: Unable to upgrade to 2.1.0

Slyrc, 2025-02-07:
I have a Turing Pi 2.4 which is currently running 2.0.5

If I upgrade over ota to 2.1.0 the whole software bricks and interestingly shows version 1.0.0 on the website.
After that basically it is useless. I can not connect via ssh anymore. 

I also flashed an SD card with 2.1.0 -> Same result here as well. Stuck and won't go anywhere. Only difference here is that I can not even open the webpage.

Only thing worked for me was using an SD card with 2.0.5 again to go back.

Also LED at node 1 is constantly flashing. Not sure if this was here before?

Slyrc, 2025-02-08:
Some more input here:

Here is the output of 2.0.5 starting (Even if there are some errors it still starts and I can login over ssh and the web ui)

```
U-Boot SPL 2024.01-rc1 (Nov 28 2023 - 13:50:37 +0000)
DRAM: 128 MiB
Trying to boot from sunxi SPI

U-Boot 2024.01-rc1 (Nov 28 2023 - 13:50:37 +0000) Turing Machines, Inc.

CPU:   Allwinner R528 (SUN8I)
Model: Turing Machines Turing Pi 2 BMC
DRAM:  128 MiB
Core:  53 devices, 23 uclasses, devicetree: separate
WDT:   Not starting watchdog@20500a0
MMC:   mmc@4020000: 0
Loading Environment from UBI... ubi0: attaching mtd2
ubi0: scanning is finished
ubi0: attached mtd2 (name "ubi", size 127 MiB)
ubi0: PEB size: 131072 bytes (128 KiB), LEB size: 126976 bytes
ubi0: min./max. I/O unit sizes: 2048/2048, sub-page size 2048
ubi0: VID header offset: 2048 (aligned 2048), data offset: 4096
ubi0: good PEBs: 1016, bad PEBs: 0, corrupted PEBs: 0
ubi0: user volume: 3, internal volumes: 1, max. volumes count: 128
ubi0: max/mean erase counter: 7/5, WL threshold: 4096, image sequence number: 0
ubi0: available PEBs: 449, total reserved PEBs: 567, PEBs reserved for bad PEB handling: 20
Read 65536 bytes from volume uboot-env to 47d36c00
OK
In:    serial@2500c00
Out:   serial@2500c00
Err:   serial@2500c00
Net:   eth0: ethernet@4500000
gpio: pin PG13 (gpio 205) value is 0
gpio: pin PG13 (gpio 205) value is 0
Setting bus to 0
Error writing the chip: 37812248
Error writing the chip: 40957960
Error writing the chip: 40957960
Error writing the chip: 40957960
Error writing the chip: 40957960
Error writing the chip: 40957960
Error writing the chip: 40957960
Button key1 state: off
Hit any key to stop autoboot:  0 
Boot device: ubi
Boot mode: normalboot
Scanning ubi 0:rootfs...
Running U-Boot script: /boot/boot.scr.uimg
```

With SD card and firmware 2.1.0 I am greeted with that while installing:

```
[    4.728315] i2c i2c-0: mv64xxx: I2C bus locked, block: 1, time_left: 0
[    6.968320] i2c i2c-0: mv64xxx: I2C bus locked, block: 1, time_left: 0
[    6.974905] realtek-smi 0-005c: failed to read chip id and version: -110
[    6.981655] realtek-smi 0-005c: unable to detect switch
[    9.048329] i2c i2c-0: mv64xxx: I2C bus locked, block: 1, time_left: 0

 _____ _   _ ____  ___ _   _  ____
|_   _| | | |  _ \|_ _| \ | |/ ___|
  | | | | | | |_) || ||  \| | |  _
  | | | |_| |  _ < | || |\  | |_| |
  |_|  \___/|_| \_\___|_| \_|\____|

This utility will perform a fresh installation of the Turing Pi 2 BMC firmware.

Note that this will ERASE ALL USER DATA stored on the Turing Pi 2 BMC, thus
restoring back to factory defaults. Do NOT proceed unless you have first backed
up any files that you care about!

If you wish to confirm the operation and proceed, either:
1) Type 'CONFIRM' at the below prompt
2) Press one of the front panel buttons (POWER or RESET), or the KEY1 button on
   the Turing Pi 2 board itself, three times in a row

If you are here in error, please remove the microSD card from the Turing Pi 2
board and reset the BMC.

Type "CONFIRM" to continue: [   11.208330] i2c i2c-0: mv64xxx: I2C bus locked, block: 1, time_left: 0
[   13.288321] i2c i2c-0: mv64xxx: I2C bus locked, block: 1, time_left: 0
[   15.368322] i2c i2c-0: mv64xxx: I2C bus locked, block: 1, time_left: 0
[   17.448321] i2c i2c-0: mv64xxx: I2C bus locked, block: 1, time_left: 0
[   19.528318] i2c i2c-0: mv64xxx: I2C bus locked, block: 1, time_left: 0
[   21.608328] i2c i2c-0: mv64xxx: I2C bus locked, block: 1, time_left: 0
CONFIRM
 |    23.688360] i2c i2c-0: mv64xxx: I2C bus locked, block: 1, time_left: 0
 | Installing BMC firmware (2s)
 /    27.848360] i2c i2c-0: mv64xxx: I2C bus locked, block: 1, time_left: 0
 / Installing BMC firmware (1s)
 / Installing BMC firmware (2s)
 / Installing BMC firmware (2s)
 - Installing BMC firmware (4s)
 - Installing BMC firmware (5s)
 \ Installing BMC firmware (7s)
[+] DONE: Please remove the microSD card and reset the BMC.
[   42.408338] i2c i2c-0: mv64xxx: I2C bus locked, block: 1, time_left: 0
[   44.488322] i2c i2c-0: mv64xxx: I2C bus locked, block: 1, time_left: 0
[   46.568321] i2c i2c-0: mv64xxx: I2C bus locked, block: 1, time_left: 0
[   48.648321] i2c i2c-0: mv64xxx: I2C bus locked, block: 1, time_left: 0
[   50.728319] i2c i2c-0: mv64xxx: I2C bus locked, block: 1, time_left: 0
[   52.808318] i2c i2c-0: mv64xxx: I2C bus locked, block: 1, time_left: 0
[   54.888320] i2c i2c-0: mv64xxx: I2C bus locked, block: 1, time_left: 0
```

svenrademakers, 2025-02-11:
It seems like you have some hardware issues; more specifically I2C-0 bus seems to be unresponsive. This part of the hardware got enabled on v2.1.0 so it explains why downgrading to 2.0.5 works. I would chech if you can reach out to turingpi on their website and see if they can do anything for you

svenrademakers, 2025-02-11:
do you have any usb cables attached to the board? its worth a shot to completely take of the power off the board with all peripherals detached.

Slyrc, 2025-02-11:
Good to know. I will try to reach out to them. Thanks!

Slyrc, 2025-02-11:
Well I removed everything from the board including the battery. Waited a few minutes and the tried again. Seems like I am not lucky here.

Slyrc, 2025-02-11:
Do you maybe know where this part is located on the board? Maybe it can be checked and repaired/replaced?

sachasmart, 2026-02-21:
> @sachasmart - yes sadly problem still persists. I ended up buying a new one.

That's a shame but thank you for following up. I've downgraded back to 2.0.5 and looks like will stay here for a while