turing-machines/BMC-Firmware #250: Latest Release not mirrored to firmware.turingpi.com

Spiker985, 2025-06-30:
As of writing (2025-06-30) [v2.1.0](https://github.com/turing-machines/BMC-Firmware/releases/tag/v2.1.0) from 2025-02-05 is not available on firmware.turingpi.com. All documentation on https://docs.turingpi.com refers to using firmware.turingpi.com as the unified location for updated firmware downloads.

Please upload v2.1.0 to your mirror site. I'd also recommend setting up a continuous deployment process so that this cannot be missed in the future, or alternatively redirect users to this GitHub's releases page if that is what you decide to do.

[image]
[image]
[image]
[image]

I only found out that there was an updated firmware version because I _happened_ to browse the tech-support channel in the discord, and noticed someone was having issues with a `2.0.5 -> 2.1.0 upgrade`. If I was to continually reference your mirror as instructed, I would have continue to be unaware that there was a newer version.

booth-f, 2026-01-07:
January 7th of 2026 and this issue is still causing pain in the developer community.

I had no idea you guys even released anything past 2.0.5. I updated back in January of last year and didn't expect you to suddenly drop another version the very next month only to then lack any new releases for another year. It's been a long year and of course I was looking at firmware.turingpi.com and never saw the updates. 

I came here specifically to find anything related to the network module on the board because I noticed when the internal switch loses external connection even local connections to other compute modules on the same board will lose network access. It's just a really bad design to turn off ALL internal network I/O if the network link is lost. 
I've had issues with iscsi targets losing access to their storage (node 4 to node 3 for the SATA ports) when network would drop but the board itself would remained powered up. I'll happily update this comment once I have a chance to apply the update and see if internal networking is resolved when external network connectivity is lost.

sudomateo, 2026-01-12:
> I had no idea you guys even released anything past 2.0.5.

Same. I've been checking https://firmware.turingpi.com/turing-pi2/ instead of this GitHub repository and I was starting to think the TuringPi firmware wasn't actively being worked on. Glad to see there's a v2.1.0 but someone should probably get the mirror site updated since, as others already mentioned, it's mentioned throughout the documentation.